starksbe/ISU-ECE-5508-S23 #1: Select FPGA board to use for project.

starksbe, 2023-01-12:


Roomybear, 2023-01-15:
From email conversation it sounds like we are using the BASYS 3 board? Just want to ensure we are set on this. If so it sounds like task #3 and #4 should be complete.

starksbe, 2023-01-15:
Yes, I believe we have decided on the board out of convenience. However, we should set aside time before or after class to formally mark these issues as closed. I'm a fan of the scrum method for these types of projects, so I'll bring it up in class this Tuesday to mark the issues as completed. This way everyone is up to speed on the state of the project. I'll create some backlog issues to get us a meeting to look at the boards and figure out how we can access them since there are already issues for defining the project scope and such.

Thank you for making sure we're moving forward with our issue management! It's helpful to have a whole team inquiring about the state of the e repository.

Roomybear, 2023-01-17:
I’m doing the best I can. I’ve never used Github but am sure excited to get used to it!   Sent from Mail for Windows From: Brandon StarksSent: Sunday, January 15, 2023 1:06 PMTo: starksbe/ISU_ECE5508S23Cc: Roomybear; CommentSubject: Re: [starksbe/ISU_ECE5508S23] Select FPGA board to use for project. (Issue #1) Yes, I believe we have decided on the board out of convenience. However, we should set aside time before or after class to formally mark these issues as closed. I'm a fan of the scrum method for these types of projects, so I'll bring it up in class this Tuesday to mark the issues as completed. This way everyone is up to speed on the state of the project. I'll create some backlog issues to get us a meeting to look at the boards and figure out how we can access them since there are already issues for defining the project scope and such.Thank you for making sure we're moving forward with our issue management! It's helpful to have a whole team inquiring about the state of the e repository.

closed 2023-01-21 by starksbe