starksbe/ISU-ECE-5508-S23 #5: Determine FPGA project deliverable(s).

starksbe, 2023-01-14:


starksbe, 2023-01-17:
I believe a group project would be fun and worthwhile. FPGAs are great for fast communication systems, so we could design our own transmitter and receiver modules using a simple BPSK modulated waveform. I could provide expertise in guiding the team since I have modeled a BPSK waveform in software.

This project would require pulse shaping, filtering, level conversion, and integration, which has a mixture of mathematical prowess from algebra to calculus. It also would have challenges that could be split into non-trivial work for individuals to be combined at the end for a desired result.

starksbe, 2023-01-17:
Another project that we could attempt is to implement a CRC code.

persjere, 2023-01-17:
The project sounds interesting. That said, I don't have a background in communications outside of completing EE4418 a few semesters ago. Because of my limited background, I will have to defer to your and Dr. Kantabutra's judgment on whether the scope and complexity would be appropriate for the course. I like that this could easily be split into subsystems so that each person could complete a manageable portion of the project.

While I understand the purpose of a CRC, I don’t know what implementing the project would entail. It would be an interesting project, but I prefer the modular design of the BPSK transmitter and receiver project.

I think that you (@starksbe) mentioned that you have access to a digital textbook to supplement the course material. I would like to start going through the material. What is the title? How do I get access to the book?

starksbe, 2023-01-17:
@persjere we could also very easily add a CRC code on top of our BPSK waveform which I think would help you feel a bit more comfortable with tackling the transceiver project. The data we would transmit would just contain a few redundant bits for the CRC polynomial to run against to see if bits were incorrectly transmitted.

I'll send you a copy of the textbook along with everyone else.

Roomybear, 2023-01-19:
I am fine with either of those projects. If we do pursue the BPSK it would be nice to have you as a lead since you know which direction to head in. Since I'm fairly unfamiliar with FPGA's just learning through the book you shared and doing a project where we have some headway sounds great to me.

Roomybear, 2023-01-28:
Also, I'm not sure how detailed you want the deliverables section when it comes to this issue, maybe Dr. Kantabutra can give us some guidance on what would be acceptable in terms of project points. Since the project is 30% of our grade I was thinking maybe a detailed proposal of the project maybe 10%, a final demonstration 10%, and then a group paper of the work/description for the remaining portion. Let me know what you guys want to do. We could do something else such as major milestones each worth 5% and then a final demonstration. Just spouting off ideas here.

starksbe, 2023-01-31:
I think this week will be a good time to come up with a system diagram of the project so that we can start creating tasks. I have a better idea of how the design will look, we just need to draw it out so we can all understand it.

It's good to know that this is not a trivial project either. We may need to keep some things simple enough that we don't create to difficult a system for us to build and make additions to. Since everyone has given their availability, I'll create a tentative zoom meeting for this week.

Roomybear, 2023-02-01:
Sounds good to me, if you want to we could meet Friday (that’s the only realistic timeframe for me). Sent from Mail for Windows From: Brandon StarksSent: Tuesday, January 31, 2023 10:57 AMTo: starksbe/ISU_ECE5508S23Cc: Roomybear; MentionSubject: Re: [starksbe/ISU_ECE5508S23] Determine FPGA project deliverable(s). (Issue #5) I think this week will be a good time to come up with a system diagram of the project so that we can start creating tasks. I have a better idea of how the design will look, we just need to draw it out so we can all understand it.It's good to know that this is not a trivial project either. We may need to keep some things simple enough that we don't create to difficult a system for us to build and make additions to. Since everyone has given their availability, I'll create a tentative zoom meeting for this week.

starksbe, 2023-03-16:
Proposal has already outlined deliverables.

closed 2023-03-16 by starksbe