starksbe/ISU-ECE-5508-S23 #7: Schedule Time to View FPGA Boards

starksbe, 2023-01-17:


starksbe, 2023-01-21:
I think since we were able to grab the boards, there is no need to meet in person to view the boards since having them is a much better substitute. If we can get a response from everyone on this thread, I'll mark this issue as done. Otherwise, I'll bring this issue up in class and we can mark it done there.

I'll be creating a new issue to formally select a time for us to meet together to discuss details and progress of the project so having availabilities in this thread will be helpful.

I am available any time after 16:00 on Mon, Tues, Thurs, Fri, Sat.

nataliacastaneda, 2023-01-24:
Sounds good, just wanted to report that I have picked up my board.

closed 2023-01-26 by starksbe