starksbe/ISU-ECE-5508-S23 #8: Setup Zoom Meeting for Planning & Progress Reporting

starksbe, 2023-01-21:


starksbe, 2023-01-21:
We need to find a time at least once per week that we can meet for fifteen to thirty minutes for planning and progress reporting. The goal is to break up the project into smaller tasks that we can accomplish each week to two weeks, not to pressure each other to work faster. If done right, the team should be able to tackle smaller issues in a convenient, non-invasive fashion.

persjere, 2023-01-23:
For most weeks, my schedule is open between 11 am and 11:45 pm on Monday/Wednesday, after 3:30 pm Monday through Thursday, or between 8 am and 5 pm on Friday.

nataliacastaneda, 2023-01-24:
Every weekday besides Tuesday I can be available between 2 pm and 5 pm.

Roomybear, 2023-01-28:
I'm available all day Friday and on th weekend.

closed 2023-02-26 by starksbe